carrefinho/forager #1: Question: Do you have any plans for making a 4x6 version of this?

csntr, 2024-10-04:
As said in the title, I really love the look, and I would immediately build a 4x6 version of it. Did you have any plans of creating one?

carrefinho, 2024-10-15:
I have some ideas for a small revision; six column could be on the cards for that. Although with the controller position it has to be a separate PCB design and can't be a break-away column like other boards.

t0bybr, 2024-10-20:
What do you plan for the revision? I really want to build this as it's beautiful and smart. I also love the case details like the reset button. I'm wondering if I should wait.

AMoynahan89, 2024-10-30:
Beautiful build, this is the keyboard I was going to build from scratch so thanks for doing all the hard work. How is the battery life on it and would it be possible to fit a bigger battery in that case or is it pretty full?