carrefinho/forager #2: Question: How long is the battery life?

aymane-eljerari, 2024-12-16:
I'm excited to build this, and im wondering how long the battery lasts. What's the largest battery I could fit in the case?

carrefinho, 2024-12-16:
One to two weeks on the left side (central), two to three months on the right side (peripheral). It's designed around the commonly available 110mAh 301230 battery, and because of the compact footprint, there is only a couple millimeters of extra room.

lukeberry99, 2025-01-01:
Out of curiosity, would this work having a third microcontroller as a “dongle” that is permanently plugged in, giving both halves two to three months battery life?

carrefinho, 2025-01-01:
@lukeberry99 
Yes, that is an option. There is [work-in-progress documentation in the ZMK docs](https://deploy-preview-2525--zmk.netlify.app/docs/development/hardware-integration/dongle).
Also check out [Prospector](https://github.com/carrefinho/prospector) for a fancy dongle.

lightningspirit, 2025-03-11:
@carrefinho Would a LiPo 301430 (32x14x4) which is 150 mAh fit in this case?

lightningspirit, 2025-05-13:
> Would a LiPo 301430 (32x14x4) which is 150 mAh fit in this case?

If anyone is wondering, I was able to use this battery in the case successfully.