carrefinho/forager #5: Feature: integrated trackpoint module

artsbentley, 2024-12-24:
Hi, first of all thank you for all the effort you clearly put into this board, it looks super clean. 

I was wondering if there are any plans to implement trackpoint module integration in the future. I'm currently quite happy with my Totem as main keyboard, but the one thing missing is integrated mouse pointer, which would be amazing on a wireless board.

carrefinho, 2024-12-26:
Trackpoints are a bit tricky because there are many different types of modules out there with no clear go-to, they require custom keycaps, and have rather high power consumption.

A trackpad or small trackball are more likely options if Forager would integrate a pointing device in the future.

twop, 2025-03-01:
Im actually curious about integrated trackpad module (possibly the very small one)